aly-fly/EleksTubeHAX #125: [Hardware] Elekstube v1 - DS3231 RTC backup batteries exhausted after only several weeks

bitrot-alpha, 2025-03-30:
I feel like my Elekstube v1's RTC battery is dead all the time.

I remember when I first purchased my clock years ago, it ran for a few weeks before the original battery died and RTC time was lost after power down. I've since replaced the battery a few times from a bulk pack of Panasonic CR1616s.

Shouldn't a "clock backup battery" last more than 3 months? I think either my coin cells are either counterfeit (purchased from Amazon), the designers should have put in a larger battery slot, or my clock has an issue. I looked up the capacity of CR1616 coin cells and it seems to be around 60 mAh. With the power draw of the DS3231 at 3 uA for timekeeping (according to datasheet), that should put the runtime for a CR1616 at a little less than 20 thousand hours or several years.. 

My hypothesis is that the Elekstube designers are actually powering the RTC with the battery all the time. That would put the power consumption at 150 (max) uA according to the datasheet. This would be more consistent with the run times I'm seeing (assuming my batteries aren't fake..). 

I'll take a look at the board again when I have time and see if the DS3231 is setup "right." I've done the 3 volt mod on the CH341 UART, but I don't think that changes anything with the RTC.

aly-fly, 2025-03-30:
Hi, how do you know that battery is dead? I don't remember, but I think we don't have any battery monitoring built in...
My Original EleksTube clock is powered 24/7 and connects to NTP on every boot, so maybe my battery is also dead, just that I have no idea :)

Martinius79, 2025-03-30:
I checked with my Elekstube Original if the RTC is powered and gives the right timestamp.
Everything seems to work fine.

```
23:12:33:805 -> Clock start...Loaded Clock config is invalid, using default.  This is normal on first boot.
23:12:33:811 -> Update from NTP Server...
23:12:34:006 -> NTP Data: 24 03 06 E6 00 00 02 A7 00 00 00 0D 0A 43 08 B6 EB 94 32 1B 4A E1 E9 F3 00 00 00 00 00 00 00 00 EB 94 32 41 24 85 DB C1 EB 94 32 41 24 8A 6B 9C.
23:12:34:017 -> NTP time = 21:12:33
23:12:34:020 -> Clock:syncProvider() entered.
23:12:34:026 -> Try to get the actual time from NTP server....Done!
23:12:34:031 -> NTP time = 21:12:33
23:12:34:031 -> NTP, RTC, Diff: 
23:12:34:034 -> 1743369153
23:12:34:034 -> 1743369152
23:12:34:037 -> 1
23:12:34:037 -> RTC time is not valid, updating RTC.
23:12:34:039 -> RTC set OK!
23:12:34:039 -> RTC is now set to NTP time.
23:12:34:045 -> RTC time = 1743369153
23:12:34:048 -> Using NTP time!
23:12:34:048 -> Done!
---
23:12:34:284 -> DEBUG: Sent availability: EleksTubeHAXClock_Elekstube/status online
```

So the RTC gives a value back, so battery seems to be powered while offline.

My personal experience with button batteries are mixed...I once received a 10x blister from Amazon and ALL of them was empty, even if the "use-before" date was years away and it was a "known brand"...Funny part was, that Amazon wanted me to send back all 10 opened pieces...

My best guess is, that you have a really bad charge of batteries...

I updated the code to have a bit more output.
Fixed also the RTC-NTP time sync a bit (NTP server call takes some time, so recall the RtcGet() function afterwards).
see https://github.com/Martinius79/EleksTubeHAX/tree/TryoutsRTCStuff

I will try to use the RTCLib or another lib with direct DS3231 support, because then the power should be checkable...

Off-Topic: 
I also modified the WiFi settings, to check if RTC is used, if the clock is offline.
I detected, that there is something wrong with the RTC get and set on the NovelLifeSE. 
Will check this out as well.

bitrot-alpha, 2025-03-31:
I did make sure to check the voltage of the new battery before I put it in last time, but that doesn't mean that it couldn't be a fake (lower capacity) battery..

The way I know that the battery is dead (without plugging the clock into PC serial terminal) is the clock resets to roughly 17:00:42 when power cycling it before NTP sync. This behavior is consistent. I know this because I occasionally have to power cycle the clock when my WiFi drops out and it fails to reconnect (most likely not an issue with the clock firmware).

Martinius79, 2025-04-07:
I created a branch where I am using the RTCLib.
https://github.com/Martinius79/EleksTubeHAX/tree/TryoutsRTCStuff

The register flag for "lostPower" of the DS3231 can be read out with it. 
If a power lost  to the RTC chip is detected, the register is reset by calling adjust() with a fictiual default time. The RTC should be then set to the NTP time.

Next run you can then see, if there was really a "power loss" detected by the RTC chip...

At least so you can see, what the RTC chip "thinks" about the battery ;)

For me, there is no problem with it in the moment...

I will try to remove the battery in the next days and see, what is happening. I also will try to find an "empty" one.

if DEBUG_OUTPUT_RTC is defined, the output should look like this on a normal run:

```
01:30:57:205 -> Clock start...DEBUG_OUTPUT_RTC: RtcBegin() for DS3231 RTC entered.
01:30:57:210 -> [  3733][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
01:30:57:220 -> DEBUG_OUTPUT_RTC: RTC found!
01:30:57:220 -> DEBUG_OUTPUT_RTC: RTC power lost = 0
01:30:57:227 -> Update from NTP Server...
01:30:57:323 -> NTP Data: 24 02 06 E7 00 00 00 FE 00 00 00 06 4F 85 2C 92 EB 9E DE 6E C7 1B EF E0 00 00 00 00 00 00 00 00 EB 9E DE B0 DA 04 21 71 EB 9E DE B0 DA 05 3F 14.
01:30:57:334 -> NTP time = 23:30:56
01:30:57:337 -> Clock:syncProvider() entered.
01:30:57:339 -> DEBUG_OUTPUT_RTC: RtcGet() for DS3231 RTC entered.
01:30:57:345 -> DEBUG_OUTPUT_RTC: DS3231 RTC now.unixtime() = 1744068656
01:30:57:348 -> Try to get the actual time from NTP server....Done!
01:30:57:353 -> NTP time = 23:30:56
01:30:57:356 -> DEBUG_OUTPUT_RTC: RtcGet() for DS3231 RTC entered.
01:30:57:359 -> DEBUG_OUTPUT_RTC: DS3231 RTC now.unixtime() = 1744068656
01:30:57:364 -> NTP  :1744068656
01:30:57:367 -> RTC  :1744068656
01:30:57:367 -> Diff: 0
01:30:57:370 -> Using NTP time!
01:30:57:370 -> DEBUG_OUTPUT_RTC: RtcGetTemp() for DS3231 RTC entered.
01:30:57:375 -> DEBUG_OUTPUT_RTC: DS3231 RTC temperature = 29.75 °C
01:30:57:381 -> DEBUG_OUTPUT_RTC: DS3231 RTC temperature is valid!
```

and if a powerlost was detected:

```
01:27:40:988 -> Clock start...Loaded Clock config is invalid, using default.  This is normal on first boot.
01:27:40:988 -> DEBUG_OUTPUT_RTC: RtcBegin() for DS3231 RTC entered.
01:27:40:990 -> [  3824][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
01:27:41:002 -> DEBUG_OUTPUT_RTC: RTC found!
01:27:41:002 -> DEBUG_OUTPUT_RTC: RTC power lost = 1
01:27:41:007 -> DEBUG_OUTPUT_RTC: RTC reports power was lost! Setting time to default value.
01:27:41:017 -> Update from NTP Server...
01:27:41:108 -> NTP Data: 24 02 06 E8 00 00 02 67 00 00 00 10 C1 4F ED 0E EB 9E DD DD 4A D8 7A F3 00 00 00 00 00 00 00 00 EB 9E DD EC A5 72 ED 33 EB 9E DD EC A5 8A 6E D6.
01:27:41:119 -> NTP time = 23:27:40
01:27:41:122 -> Clock:syncProvider() entered.
01:27:41:125 -> DEBUG_OUTPUT_RTC: RtcGet() for DS3231 RTC entered.
01:27:41:130 -> DEBUG_OUTPUT_RTC: DS3231 RTC now.unixtime() = 1672531200
01:27:41:133 -> Try to get the actual time from NTP server....Done!
01:27:41:139 -> NTP time = 23:27:40
01:27:41:144 -> DEBUG_OUTPUT_RTC: RtcGet() for DS3231 RTC entered.
01:27:41:144 -> DEBUG_OUTPUT_RTC: DS3231 RTC now.unixtime() = 1672531200
01:27:41:150 -> NTP  :1744068460
01:27:41:153 -> RTC  :1672531200
01:27:41:153 -> Diff: 71537260
01:27:41:155 -> RTC time is not valid, updating RTC.
01:27:41:158 -> DEBUG_OUTPUT_RTC: RtcSet() for DS3231 RTC entered.
01:27:41:164 -> DEBUG_OUTPUT_RTC: Attempting to set DS3231 RTC to: 1744068460
01:27:41:169 -> DEBUG_OUTPUT_RTC: DS3231 RTC time updated.
01:27:41:172 -> RTC is now set to NTP time.
01:27:41:175 -> DEBUG_OUTPUT_RTC: RtcGet() for DS3231 RTC entered.
01:27:41:180 -> DEBUG_OUTPUT_RTC: DS3231 RTC now.unixtime() = 1744068460
01:27:41:183 -> RTC time = 1744068460
01:27:41:186 -> Using NTP time!
01:27:41:189 -> DEBUG_OUTPUT_RTC: RtcGetTemp() for DS3231 RTC entered.
01:27:41:191 -> DEBUG_OUTPUT_RTC: DS3231 RTC temperature = 29.75 °C
01:27:41:197 -> DEBUG_OUTPUT_RTC: DS3231 RTC temperature is valid
```

@bitrot-alpha Maybe you are willing to try out the branch with your device and tell us the results...

bitrot-alpha, 2025-04-12:
I've edited my first comment with proper values. I was mistaken as to the type of battery in the Elekstube. It's actually a CR1220 which is smaller than a CR1616.

I was going to attempt a battery drain test, but the max sustained load the CR1220 is rated for is 0.1ma, which would put the estimated run time at several weeks for a full genuine battery. I fear if I go over the maximum load it wouldn't actually be an accurate test since the cell would be overstressed.

I will take some current draw values from my clock when I change the battery again (soon). Will report back with those results.

@Martinius79 thanks for the debug work, but it would be more useful if the battery status was displayed on one of the clock's LCDs. Maybe this could be an eventual firmware feature?

bitrot-alpha, 2025-04-12:
Well, it seems the DS3231 is drawing quite a bit from the battery. According to my multimeter, it's pulling about 400ua, which is over 4x the maximum rated constant load (0.1ma or 100ua) for the CR1220..

This load doesn't seem to vary much, whether or not the clock is powered. 

[image]

bitrot-alpha, 2025-04-12:
Looking once again at the DS3231's datasheet, this value of ~400ua corresponds to temperature conversion, which is only supposed to happen immediately after power up, and then every 64 seconds afterwards. I also find it strange that the DS3231 is pulling significant current from the battery even while Vcc is active..

aly-fly, 2025-04-12:
I suggest comparing the typical / recommended wiring diagram of DS3231 with the actual connections on the clock PCB. In last years lots of products are copy-pasted from somewhere else and each copy adds at lest one mistake...

bitrot-alpha, 2025-04-12:
I suspect my clock's DS3231 may either be counterfeit or it has ESD damage. The battery pin goes straight to the coin cell slot positive pin, Vcc goes to +3.3V, and SDA/SCL must be hooked up properly to the ESP32 since the ESP receives a time value from the DS3231.

I haven't checked ALL the pins today, but going from memory when I opened up the clock before, the pinout from the datasheet seemed to match the board.

We should also check that we're not constantly forcing a temperature conversion via the control register. Since temperature conversion already happens automatically every 64 seconds, CONV should not be manually set.

[image]

aly-fly, 2025-04-14:
Updated code is committed to Main.
Can you check if this changes any behavior of your clock?

bitrot-alpha, 2025-04-17:
I've tried the new firmware and my DS3231 still pulls a lot more current than it's supposed to. I suppose I could buy another one and replace the one on my Elekstube..

Preview of the output of a pending PR with additional DS3231 code to read from registers the Adafruit RTC library does not provide getter functions for.
```
[ 21501][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
DEBUG_OUTPUT_RTC: DS3231/DS1307 RTC found!
Square Wave output status: 0x1c
32KHz output status: Enabled
Read from control and status registers: Success
DS3231 Control Register:
EOSC:0 BBSQW:0 CONV:0 RS2:1 RS1:1 INTCN:1 A2IE:0 A1IE:0
DS3231 Status Register:
OSF:0 EN32Khz:0 BSY:0 A2F:0 A1F:0
Forcing temperature conversion now.
Temperature conversion busy flag: Set!
Waiting 2 seconds for temperature conversion to finish.
DS3231 Temperature: 26.750000 C
```

The temperature given does match my current room temperature within the tolerances of the DS3231 (+/- 3 C). It is not a constant value as far as I can tell (it has changed sometimes between reboots of the clock).

I would appreciate it if you all could post your output once I submit the PR.

bitrot-alpha, 2025-04-17:
Well, assuming I didn't mess up any bit shifting, according to the log, my DS3231 should not be consuming anywhere near the current it consumes when a battery is connected. That being since battery-backed square wave output is disabled and CONV is not manually set outside of my own code.

I guess the easiest solution for me is just to not use a battery at all, since the clock is primarily NTP based anyway.

aly-fly, 2025-04-17:
Some models don't even have a RTC chip and always use NTP on startup.
RTC is useful if you have power glitches, but WiFi is unavailable (for example some network systems need up to 10 minutes to boot or there is disruption on the ISP's network).

Those chips should be cheap, in your position I would replace it with a new genuine chip.

bitrot-alpha, 2025-04-17:
Unfortunately, I am a US citizen and the actions of my country's current president have caused the prices of electronics, especially imported components to rise..

There are modules intended for breadboarding/Raspberry Pi use available for $3 from JameCo, but I'm skeptical as to whether the ICs used are genuine. Another issue is that I would have to desolder the chip from the module. Since the accuracy of the IC could be affected by high temperatures used in solder rework, the manufacturer does not recommend doing it more than twice to the IC. Buying a genuine DS3231SN IC by itself from an electronics supplier like DigiKey or Mouser is pretty expensive for me now, at $14 plus shipping.

Other than killing batteries, the DS3231 in my EleksTube works fine. It seems to keep time well, though I suppose it's not really been tested long term since it is reset nightly with NTP time. I think I'll leave it in the clock for a while longer until I have other projects needing an RTC chip.

Hopefully my musings are useful to others who may have a similar issue.