aly-fly/EleksTubeHAX #128: Second I2C bus

aly-fly, 2025-04-08:
> I am working on a fix for the RTC support of the NovelLifeSE...The manufacturer used a second I2C bus to connect the RTC chip...So I need to modify a lib, because by default the code is only supporting one I2C bus...
> Is it OK, to place a modified lib in the /lib dir of the pio folder? It is working, but it is some kind of "fork" then...Don't know how this I then with the IP and the license stuff...We are already using some "modified" code (for NTP i.e.) and maybe it is a good idea to "extract" it and use a modified version of the lib also...

 _Originally posted by @Martinius79 in [#123](https://github.com/aly-fly/EleksTubeHAX/issues/123#issuecomment-2779070419)_

Martinius79, 2025-04-08:
**Subject: Implement RTC Handling Improvements and Fixes**

I've implemented several improvements and fixes for the Real-Time Clock (RTC) handling.

See the changes in this branch: https://github.com/Martinius79/EleksTubeHAX/tree/FixSomeRTCStuff

Here's a summary of the changes:

  * **RTC R8025T support for the NovelLifeSE clock:** This includes a modified library to accommodate the second I2C bus required by this clock.
  * **Using RTCLib for DS3231 and DS1307 clocks (EleksTube 1st and 2nd Gen)**
  * **Removed the DS1307 library from Paul Stoffregen:** This change aligns with the use of RTCLib for all supported DS series clocks.
  * **Added additional debug messages for RTC handling:** These messages should provide more detailed information during troubleshooting.
  * **Added "RTC power loss reset" for DS3231:** This feature aims to handle scenarios where the DS3231 loses power.
  * **Changed the order of the RTC set and get in combination with the NTP.**
  * **Removed some unused code**
  * **Slightly modified the GLOBAL_DEFINES.h for NovelLifeSE**

**Off-topic:** Increased the delay for the dot printing in the `loop()` function when debug output is active (now 2ms instead of 1ms).

**Next Steps:**

I need to thoroughly test these changes on all supported clock types before creating a Pull Request. I am currently unsure if the DS1307 also has the "power loss" functionality similar to the DS3231 and will investigate this during testing.

Martinius79, 2025-04-13:
I checked with all clocks and it seeems to work as expected...

PR: https://github.com/aly-fly/EleksTubeHAX/pull/129

Please check.

Martinius79, 2025-04-14:
To adjust the library and PR it to the original creator's repo is not that easy. 
The `get()` method in his original code is static because he wants to use it with the `SetSyncProvider` from the time library. This requires a global instance to call it without instantiating the class. 
However, we are using a different instance, so there’s no easy way to make this "multi-instantiable." 

I removed the `static` from the methods, which is fine for us because we don’t use `get()` directly—instead, we access it via the clock class.

I implemented a working singleton mechanism, but then the library truly becomes "one instance" only. I’m not sure if this is the right approach, as it brings significant disadvantages. So I removed it...

Martinius79, 2025-04-14:
But for now, this should be working :)
Thx for the review and the notes and the merge...