aly-fly/EleksTubeHAX #133: How to rediscover Elekstube device in MQTT and in Home Assistant?

JOR200, 2025-05-02:
Hi.
Thanks a lot for a very nice and usefull project.

I have HA set up with Mosquitto MQTT broker and my Elekstube was autodiscovered a month ago.
Since then I have given the Elekstube a new name in _USER_DEFINES.h. This resulted in duplicate objects in HA.
I also reflashed the device with the newest code a few days ago.

The entities got updated with new icons, etc, like this:

[image]

The main info like hardware and firmware (in yello) was not updated.

I'm not at all familiar with MQTT and I wonder what is the best way to delete the Elekstube device from MQTT and HA without breaking anything and then automatically rediscover it?

Please advice.
Rgds.

aly-fly, 2025-05-02:
Go to HA settings -> Devices -> Mqtt integration. 
Delete the device.
Unplug and plug the clock. New device should appear with all the correct values. You may have to change the names again.

If you are not able to delete the clock, or some old data reappears after HA restart, then you have to delete old data from the MQTT broker directly. Let us know if you will need help with this.

JOR200, 2025-05-03:
Yes. I thought of that, but was not sure if the device would be rediscovered in HA.
I deleted the device, unpluged / pluged the clock and the device was nicely rediscovered.

[image]

The duplicates are gone and the messages in the HA logs like these are also gone:

> 2025-04-30 13:24:15.570 ERROR (MainThread) [homeassistant.components.light] Platform mqtt does not generate unique IDs. ID elekstube_main is already used by light.elekstube_main - ignoring light.elekstube_main_2
> 2025-04-30 13:24:15.571 ERROR (MainThread) [homeassistant.components.light] Platform mqtt does not generate unique IDs. ID elekstube_back is already used by light.elekstube_back - ignoring light.elekstube_back_2
> 2025-04-30 13:24:15.616 ERROR (MainThread) [homeassistant.components.number] Platform mqtt does not generate unique IDs. ID elekstube_pulse_bpm is already used by number.elekstube_pulse_bpm - ignoring number.elekstube_pulse_bpm_2
> 2025-04-30 13:24:15.617 ERROR (MainThread) [homeassistant.components.number] Platform mqtt does not generate unique IDs. ID elekstube_breath_bpm is already used by number.elekstube_breath_bpm - ignoring number.elekstube_breath_bpm_2
> 2025-04-30 13:24:15.618 ERROR (MainThread) [homeassistant.components.number] Platform mqtt does not generate unique IDs. ID elekstube_rainbow_duration is already used by number.elekstube_rainbow_duration - ignoring number.elekstube_rainbow_duration_2
> 2025-04-30 13:24:15.626 ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID elekstube_use_twelve_hours is already used by switch.elekstube_use_twelve_hours - ignoring switch.elekstube_use_twelve_hours_2
> 2025-04-30 13:24:15.627 ERROR (MainThread) [homeassistant.components.switch] Platform mqtt does not generate unique IDs. ID elekstube_blank_zero_hours is already used by switch.elekstube_blank_zero_hours - ignoring switch.elekstube_blank_zero_hours_2

The only message still ocurring in the logs is this one:
> 2025-05-03 10:39:34.144 WARNING (MainThread) [homeassistant.components.light] light.elekstube_main (<class 'homeassistant.components.mqtt.light.schema_json.MqttLightJson'>) does not report a color mode, this will stop working in Home Assistant Core 2025.3, please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

Thanks for advice.
This is an amazing project

Rgds.

closed 2025-05-05 by aly-fly

Martinius79, 2025-05-05:
I am confused that this warning still shows up.

grrr....I have to check this again.... ;(

JOR200, 2025-05-06:
Here's my HA details:

[image]

Martinius79, 2025-05-07:
I created a PR, to address the warnings reported.

https://github.com/aly-fly/EleksTubeHAX/pull/134

@aly-fly: Please review and approve

JOR200, 2025-05-22:
@Martinius79 I've tested latest update and everything looks great.