aly-fly/EleksTubeHAX #135: The new IPSTUBE H407

Simaky, 2025-05-11:
Hey!
Thanks for all your hard work contributed to this project.

I wonder if it is possible to integrate a custom firmware for the new IPSTUBE H407 Clock, which has one big screen and works via their mobile app.
I want to output a custom image to that screen. I mean, I want to be able to create custom watch faces with additional information (for example, room temperature from my Tuya sensor).

If it is possible to get the firmware from that and modify it, I think I could order it and spend some time with that.
But I need to know is that possible to do what I want with it before ordering. Sadly, I can't find any information about the board installed there.

Here are a couple of photos and the link to AliExpress:
[Link 1](https://www.aliexpress.com/item/1005008810961741.html)
[Link 2](https://www.aliexpress.com/item/1005008701584523.html)

<details>
  <summary>Photos</summary>
 [image]
 [image]
[image]
[image]
</details>

Martinius79, 2025-05-11:
Hello Simaky!

I hear of a model H407 for the first time. It seems to be quite new. Not even listed on the official website of the IPSTUBE guys.
I am not sure if this is a product of the same guys who make the IPSTUBE clocks...

So as long as no one has at least one of the devices at home, nobody will be able to tell what kind of microcontroller is built in there.

If it is from IPSTUBE, I would guess that they stay with the "known" circuits and still use an ESP32, but it is not guaranteed.

I ordered one for 52 Euros, I don't know when it will arrive... in the next 10 days, I guess. Then I can tell you more...

Bye
Martinius

Simaky, 2025-05-12:
@Martinius79 Thank you so much for the reply! 
I appreciate it. I'll be looking forward to your updates.

Simaky, 2025-05-15:
@aly-fly Thanks for validating!

Martinius79, 2025-05-24:
My package from China with the IPSTUBE H407 Clock finally arrived!

As aly-fly correctly noted, this clock appears to be a modified version of the "yellow board" layout. I plan to create a dedicated repository for it soon, where I’ll document all the details thoroughly.

First Impressions/Quick Specs:
Model: IPSTUBE H407

Display: 960×360 IPS LCD, housed in a heavy-duty metal case with 1 cm thick plastic glass front protection

MCU: ESP32-S3 (R8N16 variant: 8MB PSRAM + 16MB Flash)

Flashing:

No auto-download circuit – requires holding the second button during power-up to enter download mode

No read/write protection (firmware backup successful!)

Pinout: Not fully mapped yet, but custom firmware should be feasible.

More to come once I dive deeper!

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Simaky, 2025-05-24:
@Martinius79
> More to come once I dive deeper!

Wow! I really appreciate it, huge thanks!
Waiting for a new repository :)
Looks like you'll be the first one who create it for this device

aly-fly, 2025-05-25:
Nice. From the photos it seems that it's pretty quality made.

aly-fly, 2025-09-09:
Thanks for the effort. 
I already wrote my opinion before, that this device is way different than original clocks with discrete LCDs, therefore it would break the code a lot, and is not worth the effort to integrate here. 
I would use it  for something pretty different than just a clock.  In my case I have such display on the desk at work and at home, showing current rain radar image, weather forecast, lunch menu in the nearby restaurant, bitcoin price and trend, kids daily schedule and lunch in the school. Screens automatically cycling every 5-10 seconds. Very useful.

[image]